armbian/config #227: Armbian GUI password

testocpp, 2024-09-24:


testocpp, 2024-09-24:
Hi 

New to the enviroment so may be missing something obvious on my side but even after mutiple attempts I am not able to log into Armbian GUI (for the following built) for bananapi M5. GUI shows "pi" username instead of root on the green gui screen and not been able to find password for the "pi" username. Tried the usual 1234, armbian,etc password none of which worked. 

Armbian_24.8.1_Bananapim5_bookworm_current_6.6.47_xfce_desktop.img.xz

SSH seems to be disabled therefore not able to use that route and cannot change to another user (such as root) to login. 

Looking for help on how to resolve this. 

Regards

igorpecovnik, 2024-09-24:
> Looking for help on how to resolve this.

https://forum.armbian.com/

closed 2024-09-24 by igorpecovnik