armbian/config #229: Fix openhab install

moodyblue, 2024-11-04:
Openhab now uses zulu17, but the installer still mentions zulu11

EvilOlaf, 2024-11-04:
Feel free to provide a PR to fix this. Our development efforts are going to its sucessor: https://github.com/armbian/configng

igorpecovnik, 2024-11-04:
We switch to new one soon, old is unmaintained, will be archived soon.

Adding is simple: https://docs.armbian.com/User-Guide_Armbian-Software/

moodyblue, 2024-11-04:
Will there be an option in configng to install openhab ?

igorpecovnik, 2024-11-04:
> Will there be an option in configng to install openhab ?

There are no specific plans. Until end of the year we plan to stabilise what is currently in the system and accepting well tested community contributions.

Try following our instructions. I spent a day making them and they should be on the level that anyone, that knows what GitHub is, should be able to understand them. If not, I will improve them. You should need less then one hour to add OpenHab to this system when doing it first time.

See extension to the build system that generates image with preinstalled OpenHab. This works:
https://github.com/armbian/os/blob/main/userpatches/extensions/openhab.sh

igorpecovnik, 2024-11-06:
https://docs.armbian.com/User-Guide_Armbian-Software/HomeAutomation/#install-openhab

closed 2024-11-06 by igorpecovnik