armbian/config #231: No DTB available on Odroid HC1

belegdol, 2024-11-14:
Hello,

in the process of investigating the slowness of my HC1 I tried reapplying the overlay. To my surprise, none seem to be available once I go to SY017 setting. The following gets shown in the console
```
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
ls: cannot access '/*.dtbo': No such file or directory
```

igorpecovnik, 2024-11-14:
This is the fix https://github.com/armbian/configng/pull/256

belegdol, 2024-11-30:
Is it supposed to be fixed in `24.11.1.1111.090121`?

igorpecovnik, 2024-12-01:
No, it seems it doesn't work. We might have double problem here. 1st is that folder overlay got missing ... but not just here. This is more critical problem. But I am over the top with work and can't do anything for awhile.

belegdol, 2024-12-03:
https://github.com/armbian/configng/commit/6fc59d109968f790b66ae0275a75d634139ee565 is supposed to be in `25.2.0-trunk.116.1202.192010` if I understand correctly. I still get shown the "Manage device tree overlays" option shown though.

igorpecovnik, 2024-12-03:
There are several problems with dt:
https://github.com/armbian/configng/pull/299
also this is getting added:
https://github.com/armbian/configng/pull/298

igorpecovnik, 2024-12-09:
https://github.com/armbian/configng/pull/298

closed 2024-12-09 by igorpecovnik